ClusterM/fdskey #4: How to flash the bootloader

donly, 2023-06-17:
Hi, there!
I read the instructions from https://github.com/ClusterM/fdskey#flashing-the-firmware, but still get stuck in connect to target:

```
Can not connect to target!
If you're trying to connect to an STM32W1xx device, please select Normal or HotPlug mode from Target->Settings menu.
If you're trying to connect to a low frequency application , please select a lower SWD Frequency mode from Target->Settings menu.
```

donly, 2023-06-17:
When using this setting, it can be connected, but it shows an unknown device, as shown in the figure
[image]

What kind of `Mode` and `Reset Mode` should be use.

closed 2023-06-20 by donly

ClusterM, 2023-06-21:
Sorry for the slow reply. Double check the connections and make sure the UC is not broken. Also? check the power. Reset mode depends on your programmer. If it has a NRST connection - use hardware reset.

donly, 2023-06-22:
Thank you ClusterM. As I'm a newbie and don't know how to use it, it's been solved now.

partlyhuman, 2023-07-27:
> Thank you ClusterM. As I'm a newbie and don't know how to use it, it's been solved now.

@donly, I too am new to STLink, I've got a clone ST-Link V2 and using the ST-Link utility. Can you elaborate for other newbies what you did to get it working?

My issues thus far:
* Only works in Mode = Hot Plug, otherwise fails to connect
* Seems to hang up if Hardware reset mode is used and NRST are connected
* "Read out protection is activated" trying to erase or program the STM32
* "the elf loader file is not accessible" when trying to program the STM32

Thank you in advance for any advice you can share! :)

donly, 2023-07-27:
Maybe your clone ST-Link doesn’t support this chip, can you try another flash tool

partlyhuman, 2023-07-27:
> Maybe your clone ST-Link doesn’t support this chip, can you try another flash tool

It's quite possible the clone ST-Link V2 usb programmer is the issue. I don't currently have any alternative... is that what your issue was?

donly, 2023-07-27:
Yes, just forget this programmer..