ClusterM/fdskey #6: No space for beveling on the PCB

rfmerrill, 2023-06-27:
Different PCB manufacturers likely have different requirements, but for people who want to file bevel by hand on the PCB, it would be nice if there were a bigger cap after the contact fingers before the edge. As it is, it's hard to file a bevel without hitting the contacts, potentially making them more vulnerable to corrosion.

I don't want to push a lot of work on you, I just don't have Diptrace to do it myself. Maybe I can edit the gerber files directly?

ClusterM, 2023-06-30:
Hello. How big should the gap be? I'll edit it by myself, thank you.

rfmerrill, 2023-06-30:
PCBWay says minimum 1mm but for hand-beveling 2mm would probably be better.

Also you likely also want chamfer cuts at the sides but that could also be done by hand.

ClusterM, 2023-07-18:
Added 1.5mm.

closed 2023-07-20 by ClusterM