ClusterM/fdskey #7: Error :(  SD card not ready on small SD cards

donly, 2023-06-28:
I bought some sd cards with different capacities to test, and found that the cards with a capacity below 4g would have this error.
```
Error :( 
SD card not ready
```

PASSED: 4g, 8g, 16g, 32g, 64g
FAILED: 1g, 128m

Notes: All tests using the same file system format FAT32. 1g and 128m sd cards are low speed without C10.

ClusterM, 2023-07-20:
I'm still waiting for delivery. But can you try interim version? http://clusterm.github.io/fdskey/

donly, 2023-07-21:
Thank you for the updated, I tried the latest build, now it works with 1g SD card, but 128m still gets FAILED.

ClusterM, 2023-07-21:
> Thank you for the updated, I tried the latest build, now it works with 1g SD card, but 128m still gets FAILED.
> 

I'll wait for delivery with SD cards, thank you!

ClusterM, 2023-07-24:
Please try again: http://clusterm.github.io/fdskey/

donly, 2023-07-25:
> Please try again: http://clusterm.github.io/fdskey/

FAILED on 128m. 

It seems to take longer before the OLED lights.

ClusterM, 2023-07-29:
> > Please try again: http://clusterm.github.io/fdskey/
> 
> FAILED on 128m.
> 
> It seems to take longer before the OLED lights.

Try again please.

donly, 2023-07-29:
About 1.5 seconds the oled turn on and show the same error.
I Updated the latest `fdskey.bin` from http://clusterm.github.io/fdskey/ using a 64gb card, then insert a 128m card with fat32 of file system.

ClusterM, 2023-07-29:
Weird. Yes, i received SD cards from Ali.

[image]

And every card works fine for me. I'll try to add some debug info to the error message.

ClusterM, 2023-07-29:
Please try the current version and tell me the error code.

donly, 2023-07-30:
I tested 5 128m cards, all of them have error code `2`, except one error code is `7`.
Additional tests with different file formats for this card, the error code is still `7`, and only this card has 1.5 seconds delay to light on.
Every cards works well on windows, maybe it's the quality of the cards, after all, it's a cheap cards..
[image]

ClusterM, 2023-07-30:
Error 2 occurs at the very early stage, like SD card is not connected at all. Weird :( I have no idea.
In any case, I wouldn't recommend using these cheap SD cards because they can die and lose your save files.

ClusterM, 2023-07-30:
Maybe these SD cards can't work in SPI mode for some reason? PC card readers usually do not work via SPI.

donly, 2023-07-30:
Okay, I will use good branded cards. I'm very happy that the card you purchased is OK.
Anyway, thank you very much for your effort.

closed 2023-07-30 by ClusterM