ClusterM/fdskey #9: Error: 02

williamgentry12, 2023-07-29:
If i get the following error code: 02

Does this mean something is faulty with the FDSKey hardware, maybe broken solder joint on the STM32 chip?

Err. 02 = No disk power supply — batteries and/or AC adaptor.

ClusterM, 2023-07-29:
Hi! Can you try interim firmware version? http://clusterm.github.io/fdskey/
Put fdskey.bin file on a microSD card and hold all four buttons during power-on to update the firmware.
If problem persists it's hardware problem on MOTOR_ON pin (shorted to ground?).

williamgentry12, 2023-07-29:
Thank you, I will try :)

kingtreelo, 2023-07-30:
i was having the exact same issue and this has fixed the issue for me

williamgentry12, 2023-07-30:
works for me too

ClusterM, 2023-07-30:
I'll add this fix to release. Thank you for bugreport!

kingtreelo, 2023-08-01:
this issue seems to have come back on my device, tried flashing it again and still getting the same error

ClusterM, 2023-08-01:
> this issue seems to have come back on my device, tried flashing it again and still getting the same error

:( Please try this firmware: https://cluster.wtf/temp/FdsKey.bin

But it's not normal. Looks like some hardware problem on line MOTOR_ON.

kingtreelo, 2023-08-03:
Just tested on 2 of my Famicoms, all looks good so far, will report back once it has been powered off a while, i'll update it in a few hours

kingtreelo, 2023-08-04:
Looking good still today

ClusterM, 2023-08-05:
Great, I'll include this in the next version.

closed 2023-08-05 by ClusterM