ClusterM/fdskey #12: Programming Issue Using ST-Link-V2 - Read Out Protection Error.

PercyRosasR, 2023-10-14:
First of all, thanks for all the effort in this project, thanks to ClusterM for creating this.

I have made 2 unids following all the instructions on the FdsKey principal page, I am having some issues with the connection and programming the STM32G0B0CET6 chip.

I use the ST-Link-V2 Clone, with STM32-ST-Link-Utility, this is because is the only way I had to connect the board via stlink.

1.- Using STM32-ST-Link Utility

[image]

23:41:12 : ST-LINK SN : 700039000F0000315151524E
23:41:12 : V2J42S7
23:41:12 : Connected via SWD.
23:41:12 : SWD Frequency = 4,0 MHz.
23:41:12 : Connection mode : HotPlug.
23:41:12 : Debug in Low Power mode enabled.
23:41:12 : Device ID:0x467 
23:41:12 : Device family :Unknown device

My first question is, if this Device family information is correct?

2.- When I try to program the chip using the bootloader bin fie, this issue happens:

[image]

[image]

Pressing the ok button , the next error appears.

[image]

[image]

[image]

Do you have any clue or advice to solve this problem?

I have already try to disable the Read out protection, but I got this issue after applying.

[image]
[image]

3.- Also want to explain, why I'm not using the STM32Cubeprogrammer.

This is because when I connect my stlink32 in the combo box Serial Number, i got the letter P in both of my boards.

[image]

and when I change the Shared combo to Enable, I got this, and I am not able to connect the device in any way.
[image]

I hope you can help me with this issue.

Thanks for all in advance and greetings from Peru.

ClusterM, 2023-10-14:
Hi! The STM32-ST-Link Utility will not work :( It doesn't recognize this microcontroller (it's a new model). I don't know why STM32CubeProgrammer has such weird behavior, but you need to use it. Please try the CLI version:

```
STM32_Programmer_CLI.exe -c port=SWD --erase all --download bootloader.bin 0x08000000 --download fdskey.bin 0x08020000
```

PercyRosasR, 2023-10-14:
Hi Alexey, 

you are completely right. The old stlink software does not allow me to complete the programming(connects but gives the lock error).

The cube software has a different problem.

The solution you gave is the correct one, by cmd and the command I was able to program my chips.

Now I'm having a different issue I will open a new topic, to close this one.

Thanks in advance for your help.

closed 2023-10-14 by PercyRosasR

partlyhuman, 2024-07-23:
Thank you! Issues flashing the STM32 aborted my efforts to build the FDSkey last year. This new instruction worked perfectly for me the first time.

FYI if you're on MacOS, you can find the CLI at
```
/Applications/STMicroelectronics/STM32Cube/STM32CubeProgrammer/STM32CubeProgrammer.app/Contents/MacOs/bin/STM32_Programmer_CLI
```
assuming the default installation.