ClusterM/fdskey #13: Not SD Card problem.

PercyRosasR, 2023-10-14:
Hello Alexey and everyone.

I am having the next issue with my unit.

When I turn my fdskey without a micro sd card I got this message.
[image]

When I insert the micro sd, the screen turns black without response.

[image]

If I start the console with the micro sd inserted, the same problem shows, black screen.

I have already try with 2 micro sd, one 2gb Samsung and 16gb sandisk.

Hope you can give me some advice to solve this issue.

*I will reweld  the sd card holder pins to test it again.

Thanks in advance.

ClusterM, 2023-10-15:
Seems like the card init process has not even started. Please check the card detect line/pin. Sometimes the card detect mechanism can also be broken. Try to short the card detect line to the ground. The card socket must connect it to the ground when an SD card is inserted.

PercyRosasR, 2023-10-15:
Hi Alexey, good morning and greetings from Peru.

I shorted the cd line to the ground and I got a black screen.

I have already made 3 of the fdskey boards to test and all those have the same behavior.

What else could it be? Do you think it might be faulty micro sd card sockets?

Let me know if you need more detail, I followed all the instructions, but I don't know if could be a bad capacitor or resistor.

Thanks in advance for your help.

ClusterM, 2023-10-16:
Weird. I have no idea.

PercyRosasR, 2023-10-16:
After doing some additional tests, I was checking the voltages in the board.

And I was having less than 3.3v, in some cases 2.6v. Then I realized that I had a mod to connect USB C to give power to the console.

So I connected the regular 10v adapter and my unit came alive, so was a mistake from my last mode.
[image]

I will try to fix that, but the system is pretty good and works properly.

Thanks, Alexey for your help.

This device is pretty awesome.

ClusterM, 2023-10-16:
You are welcome :)

closed 2023-10-16 by PercyRosasR