ClusterM/fdskey #14: ROM that bypassed KYODAKU becomes Invalid ROM

Hidetyo, 2023-10-25:
Hello,

An "Invalid ROM" error occurs in the ROM of a homebrew game that has been KYODAKU bypassed.
This ROM works correctly on the emulator.

This problem occurred with the sample ROM at the URL below: 
https://github.com/bbbradsmith/NES-ca65-example/tree/fds

ClusterM, 2023-10-25:
Hello.
That's because file amount value is greater than actual file amount. I need to add some special mode for it, i think...

ClusterM, 2023-10-25:
Oh... I already fixed it, lol. Are you sure that you tried the current interim build? It should be 78424a5.

Hidetyo, 2023-10-26:
Oops, I only tried Bootloder.
I updated the main firmware and it works correctly.

My apologies! And thank you very much.

Hidetyo, 2023-10-26:
But when I use it with Duplifds(v1.1 and interim builds), I get an Inval ID Block error...

ClusterM, 2023-10-26:
> But when I use it with Duplifds(v1.1 and interim builds), I get an Inval ID Block error...

I know, I'll try to add some special mode to DupliFDS.

kazueda, 2024-02-23:
Hello from Japan.
When I am trying to bott a lunix.fds of the lng-fds project, your firmware built from a current github cloned files shows "invaild rom."
Are there some advice to overcome this situation?
Sincerely yours,

PS
Url of lng-fds in github.com
https://github.com/decrazyo/lng-fds

kazueda, 2024-02-23:
Some Progress.
File size of linux.fds is 14 kB.  So I made  65516 bytes file from the linux.fds.
But when I proceed to load shell program, "Error, Wrong block CRC" was displayed.

ClusterM, 2024-02-23:
@kazueda Hi! I'll check, but a bit later. Can you create a new issue about it? So i will not forget.

kazueda, 2024-02-23:
@ClusterM  I will set new issue.