ClusterM/fdskey #18: Some files don't seem to load with a header

GreatGlobox, 2024-05-21:
Hi there. I mainly bought this because I'm into composing chiptune and I wanted to listen to it on the real hardware with authentic FDS audio. I use FamiStudio to compose, and it's able to export either FDS or NES files. For some reason, the files exported by it aren't working on my FDSKey. At first, I thought the issue could be related to NMI not being disabled during disk loading, but then I found out that removing the 16 byte FDS header and padding the end of the file makes it load. I have no idea why this is, because other games have headers and they load just fine. I can just strip the headers for now since it works, but I just wanted to mention this in case.

GreatGlobox, 2024-05-21:
Just wanted to report back to say that it was actually an issue with the file size on the FamiStudio end. When the fds file was padded, it was only padding it to 65500, while the size of the header was supposed to be included to make it 65516. The FDSKey was assuming the smaller filesize meant no header was present, hence it failing to read the files.

closed 2024-05-21 by GreatGlobox

TakuikaNinja, 2024-11-26:
It turns out that this issue is present with the FDS homebrew/test programs on romhacking dot net. They include a header but weren't padded to 65516 bytes. Just leaving this here in case anyone else ran into this.