ClusterM/fdskey #20: Unable to load any dump

LucentW, 2024-08-09:
Hey,
I got ahold of one of the prebuilt FDSKey PCBs off AliExpress, it turns on regularly, I am able to select a dump off the SD card, it signals correctly to the RAM adapter a disk was inserted, but it indefinitely loops around the disk, the FDS BIOS staying stuck on the "now loading" screen (never passing KYODAKU/no black screen before the copyright notice).

I tried already updating to the interim build (both bootloader and software), changing microSD card, using different RAM adapters (both work perfectly fine with my FDS drive) and reflowing all components. I also checked the two resistors on the board, since testing caps while in circuit is not really a good test, and they respect the BOM (both 1.5k).

What could be the issue? Which connections should I check?

LucentW, 2024-08-12:
Figured out that the via for READ_DATA was open potentially due to a PCB manufacturing error, I just ordered another prebuilt from another seller and called it a day. Sorry for the hassle :)

closed 2024-08-12 by LucentW

ClusterM, 2024-08-12:
Hi. Sorry for not so fast answer. Ok :)