ClusterM/fdskey #23: FDSkey gives me Error 03 while attempting to save

Cyrus331, 2024-09-27:
When FDSKEY plays some games that need to be archived, a 003 error message will appear. Other games that do not require archives can be played normally. Is it a RAM problem, a FDSKEY problem, or an SD card problem?

ClusterM, 2024-09-27:
Hi! What you mean by "archived"?

donly, 2024-09-27:
I have tried using an SD card to play a game that requires a save file and it results in error 07.
Can you try with another SD card

Cyrus331, 2024-09-27:
> Hi! What you mean by "archived"?

It means this game has a save function，

Nazo no Murasamejou (1986)(Nintendo)(J)[tr En].fds

Metroid (v1.02) (1986)(Nintendo)(J)[h3].fds
[image]

kalm-traveler, 2025-08-17:
I get this when trying to load specific games as well - just got my FDS RAM adapter today and could not get Akumajo Dracula (the first Castlevania) or Metroid to save (or in Metroid's case to even load the game after trying to create a save file). ERR 03 as well. Tried the same disk images on an Everdrive and no issues. Is there a setting on the FDSKey to 'allow' writing to disk?

donly, 2025-08-19:
Where did you use it?  @kalm-traveler 

> You must pull-up write data line to VCC via resistor (5K-20K) if you want to save game progress, because Twin Famicom has no pull-up resistors inside it.